rogerclarkmelbourne/Arduino_STM32 #943: Code isn't uploaded to the bluepill

omoding-joshua, 2025-05-06:
Hey.  I need help. when I upload using the stlink, the arduino IDE says that the code was uploaded yet it didn't. what am I doing wrong?

board707, 2025-05-06:
Please show the code and the diagnostic messages of the IDE

omoding-joshua, 2025-05-06:
okay. Thanks for responding. Here is my code
void setup() {
  // put your setup code here, to run once:
Serial.begin(9600);
}

void loop() {
  // put your main code here, to run repeatedly:
Serial.println("Hello world");
delay(1000);
}

here the diagnostics
Sketch uses 12412 bytes (18%) of program storage space. Maximum is 65536 bytes.
Global variables use 2064 bytes (10%) of dynamic memory, leaving 18416 bytes for local variables. Maximum is 20480 bytes.
C:\Users\USER\AppData\Local\Arduino15\packages\stm32duino\tools\stm32tools\2022.9.26/win/stlink_upload.bat C:\Users\USER\AppData\Local\Temp\arduino_build_817233/testing.ino.bin

board707, 2025-05-06:
Printing to the Serial is not a good test for STM32 boards, because the problems with USB-UART bridge.
Try blinking with onboard LED to test the code.

stevstrong, 2025-05-06:
How do you know that code is not uploaded?
Pay attention to jumpers, both must tie the pins to GND.
setup0 and loop0 will not work.

omoding-joshua, 2025-05-06:
> How do you know that code is not uploaded?
I checked the used stm32 programmer to read the memory and it was empty

> Also you didn't show the full output of the upload process
that's all the Ide is giving me

> Printing to the Serial is not a good test for STM32 boards, because the problems with USB-UART bridge.
Try blinking with onboard LED to test the code.
okay. Let me do that tommorow

Pk-IND, 2025-05-06:
I need help. when I upload using the st link, the arduino IDE 1.8.19.  the code was not uploaded. just i put led blinking code.